PetoiCamp/OpenCat-Old #1: Add "GitHub Topics" at OpenCat main repositpory

fititnt, 2018-03-13:
GitHub topics (https://help.github.com/articles/about-topics/) is one good way to make easier to people find good projects on GitHub. Even it this repository add a better description, would be a good idea add specific topics.

Here one issue i wrote some time ago about how the good usage of topics could affect searching https://github.com/RocketChat/Rocket.Chat/issues/10032.

About that topics add, I'm not sure. But `robotics` (see https://github.com/topics/robotics) and `iot` (see https://github.com/topics/iot) at least in my opinion (consider that I'm not a I'm not a hardware developer) could be a good start

RongzhongLi, 2018-03-17:
I added a short description and tagged three topics: robotics, quadruped, opencat. 
Would this solution close the issue?

fititnt, 2018-03-17:
Sure.

From time to time it a good idea keep an open eye for similar projects on GitHub and how they use topics and some special words.

closed 2018-03-17 by fititnt