PetoiCamp/OpenCat-Old #3: Wiki tab update and collaboration info

DarkOmanRojer, 2018-04-27:
Hi,
In the Code tab there is written that info about partecipation to the project can be find in the wiki tab but there no such info. Then I tryed to open the manual of Assembling instructions (mini kit) from table of content and there is nothing there.
Can you please tell us how can we help you and how can I have the open cat and try to develop it sharing info with you?
BR,
Aliaksei

RongzhongLi, 2018-04-28:
Hi,
This repository is created for future sharing and collaborations once we have a stable product. 
I'm currently hosting a private repository with some volunteers for testing the codes. The contents will be migrated here once ready. Thank you! I hope it can happen sooner.

asmales, 2018-04-30:
Would it be possible to become a member of your testing community?
Hope to hear from you soon.

borntoleave, 2018-04-30:
@triffid Thanks for sharing the page. It's inspiring! As an unemployed foreigner who wish to devote full-time on the project, I have to get proper sponsorship on both living and visa before releasing all my technic details. I do hope to bring in more talents as soon as possible.

borntoleave, 2018-04-30:
@asmales Hi, I replied to your email shown on GitHub profile.

chaitanya11, 2018-05-01:
I do have the same question :)

Breezor, 2018-05-01:
I understand your situation, having worked in new business development.  Open sourcing is not your biggest problem as you need to sustain yourself first.   Your project seems viable.   Having you considered venture capital, or attempted to find an angel investor?  You would have to give up some portion of ownership.    Crowd funding is an option but also has its problems.  Where are you located?

 

Regards,

 

Brian Gale

borntoleave, 2018-05-02:
I'm living in Winston Salem, NC, USA. I will move to larger cities very soon. I'm going to visit some potential cities in May, but have not yet decided which city to stay.  If I cannot find sponsorship in time, I'll have to return to China.

I'll abandon my "bedroom lab" and pause any development during the coming journey. For me, the challenges are not even in the technic part. I'm getting headache on homeless, unemployment, and an expiring visa :) 

Your interests are already sufficient for me to believe in the value of my project. And I'd love to take it as my career even without external support. Crowdfunding is a path that I'm going to try in any way. 

I'm also connecting to investors. However, they're measuring the projects in a different gauge. Basically they are not confirmed of its commercial value (If you have expertise in those business calculations please teach me through rongzhong.li@petoi.com). The market value depends on the price, the price depends on the cost. It seems more practical to start with the mini version. I highly believe that with mass production, it can be affordable. Look at these similar products: https://www.robotshop.com/en/quadrapod-development-platforms.html

I'm optimizing the mini model to reduce the complexity and cost whenever possible. But if you want to buy a beta version, ironically, you'll have to pay more for less completeness. I'm still ordering from eBay and manually assembling some parts. There are a lot of people asking for the beta version, but I don't want to disappoint them with an expensive and buggy product. 

If there're any updates regarding the product release, I'll announce them on my YouTube channel and twitter(https://twitter.com/T4Petoi). 

Again, thanks for your support and patience!

chaitanya11, 2018-05-02:
@borntoleave  why cannot you try [kickstarter](https://www.kickstarter.com/)

borntoleave, 2018-05-02:
@chaitanya11 Kickstarter needs a green-card holder to register the account. While Indiegogo doesn't. My current priority is finding long-term partners who can commit full-time with me in a stable office with assembling line and warehouse. It's a hardware product. A single man is unlikely to be able to fulfill the whole kickstarter campaign orders, even if most jobs could be outsourced.

closed 2019-04-23 by borntoleave