PetoiCamp/OpenCat-Old #4: STL

STGRobotics, 2018-05-07:
I like to help, I can create a document on assembling the STL also I can convert the STL into step file.

borntoleave, 2018-05-07:
Hi,
Thanks for your offer and bringing poppy project into my attention. 

I have seen this robot somewhere. It seems to be a great open-source project for me to learn from. They also have excellent organization and documentations, which means a lot of work and resources. Do you happen to know how they manage to support their maintenance team? 

I hope to share my files as soon as my team get proper support.

STGRobotics, 2018-05-07:
Most company that are open source, use email, YouTube, web site and GitHub. If you place you file on GitHub it make support EZ. by having different version

WBWW, 2018-05-08:
Rongzhong,

Opencat is awesome, lots of people will want to help you if you work out what needs to be done. That's the spirit of open source.

I think creating a kit for people to be able to buy would also be a great idea. Beginner/intermediate people could just buy and build it. It would be good for schools as well. Indiegogo and Kickstarter are good places to look at kicking that off, once you know how much it needs to cost for it to be viable. 

But you should definitely also have a fully open source version available for independent makers who want to build it from scratch.

farmbot.io is a good example of a very successful project available in both kit and open formats.

STGRobotics, 2018-05-08:
That's awesome you want an kit version, this is going to take lots off planning, distribution, manufacturing and source code control also funding.

borntoleave, 2018-05-10:
Hello, sorry for delayed response. I'm in the week for final exams, and next week I'll visit a couple of universities. 

I'm already sharing a step-by-step instruction for the mini kit with my students and several collaborators like you. Here's a short video showing my students' work so far: https://youtu.be/T1OEnOONbIA . I really enjoyed sharing my design. I also saw their struggles and is working on some improvements. 

My plan for the summer is to find a city to stay, and finalize the mini kit before I have to leave the US. In China I can meet directly with suppliers and manufacturers, though I'll lose easy access to Google, GitHub, YouTube, etc. I could share the source codes to makers who can compile everything from scratch, but it won't generate revenue for the company. I still have to build a team to serve the public with simpler and cheaper kits, with necessary educational features. That's how my company will survive and achieve my proposed features on the full cat. 

There're other reasons that I'm not opening the sources to larger scales for now. I've received hundreds of emails/messages since I prematurely posted the demo video(it was for one potential grant and they're not confirmed with my BP). I responded to every single question and it was quite distracting. Though I hope to focus on development as soon as possible, my current priority is finding long-term partners to work closely with me. Without an efficient team and proper division of labor, too many branches will actually slow down my progress. The main purpose of my upcoming trip is to collect such a team and plan future steps.

There're also challenges from copycats and better funded competitors. I really need to plan carefully, considering that I've been raising the cat with my small salary, and will devote full-time to this career. 

Last but maybe the most important: I've been learning a lot from open source community and have the vision to make this cat open source. However, this is my first time to collaborate on a project that's getting overwhelming interests. Actually I'm not sure how things are operated in the community. Please feel free to share your experiences and comment on my thoughts. My email is rongzhong.li@petoi.com. I'll try to reply in time. I'm in New York time.

Thanks for your patience!

peterhyphenhyphen, 2018-07-08:
I’m a Technology teacher at a small high school in Perth, Western Australia. I have been involved with developing projects with Universities (the University of Western Australia Dept of Electrical and Electronic Engineering) and getting schools involved. We built robots (literally hundreds) using stepper motors scammed from 5.25” floppy drives, we made a game playing robot called Geranos  (crane in Greek), digital scales using strain gauges, digital anemometers, my school designed and made a 3m satellite dish (student built) that was our internet connection, drillbot a robot designed to drill 0.5mm holes in aluminium extrusion for our digital air track used in physics classes. Other projects included digital puppets, a floor turtle for drawing using LOGO with a dll to talk to the Step4A board we designed to let a parallel port control four nema17 steppers using eight bits. 
Between 1998 and 2010 we ran Technology projects with around 200 schools across Australia in order to get Technology teachers to take on hard fun projects using microcontrollers supported by university providing CAD drawings, electrical schematics, code (five languages), BoM sheets, naked PCBs etc. This was preArduino and Raspberry Pi so we were using simpler and older systems. Documentation etc takes ages but I am most willing to help. Our project was called Project Genesis it ran for around ten years.  http://eee.uwa.edu.au/genesis is now offline but wayback machine has some of our stuff. Registration was free for schools as was support. I want to build OpenCat and use it as the core project with lower school classes to allow design teams to focus on their team strengths and improve specific design parts. I would love to get in touch to see if this is possible. Best wishes, Peter Spicer-Wensley

borntoleave, 2018-07-08:
@peterhyphenhyphen I'm working on a new version with much simpler frame but also support Arduino/Pi. Actually I reserved enough room for other controllers as long as you can convert C/Python codes to their languages. The hardware of the new version should be ready in two months. I'm going to launch a crowdfunding campaign in five month, considering challenges other than technology. I'll send out invitations to beta testers earlier than the formally campaign. Thank you and stay tuned!

closed 2019-04-23 by borntoleave