PetoiCamp/OpenCat-Old #5: (BOM) bill of material

STGRobotics, 2018-05-08:
Here is the BOM based on the https://www.hackster.io/petoi/opencat-845129, If you continue to up your have the BOM list ready

Hardware components:
SparkFun Arduino Pro Mini 328 - 5V/16MHz × 1
https://www.sparkfun.com/products/11113

Raspberry Pi 3 Model B or B+ × 1
https://www.raspberrypi.org/products/raspberry-pi-3-model-b/
https://www.raspberrypi.org/products/raspberry-pi-3-model-b-plus/

Adafruit PCA9685 PWM & servo driver × 1
https://www.adafruit.com/product/815

Micro SD card × 1
Heat sink × 2
Compression spring × 13
Torque spring × 4
Extension spring × 1
Flat self-tapping screws (various) × 1
Rivets (various) × 1
Infrared sensor and remote × 1
Buzzer × 1
Amplifier × 1
Cellphone speaker × 1
USB microphone × 1
Capacitor × 1
Resistor (various) × 1
18650/18500 batteries × 2
Battery holder × 1
MG92B servo × 13
MG91 servo × 1
Longer servo screw × 14
Pi noir fisheye camera with lights × 1
ToF lidar × 3
Adafruit Capacitive Touch Sensor Breakout - MPR121 × 1
GY-521 MPU-6050 3 Axis Gyroscope + Accelerometer Module For Arduino × 1
Slide Switch × 1
Male/female pin connector (various) × 1
Flat washer × 14
Lock washer × 14
Right angle connector × 1
Rainbow wires × 1
USB to micro USB cable × 1
SparkFun FTDI Basic Breakout - 5V × 1
Pan/tilt holder × 1
Heat shrink tubing × 1
Electrical tape

borntoleave, 2018-05-10:
Thanks, this is an easier start for me. 

My list on Hackster is a mixture of the mini kit and the full cat. I'll group the components for building the mini kit from scratch, and note their specific parameters and requirements. You may help to find the best online links for individual makers. I'll continue improving the kit for scale production and contact wholesale suppliers for larger orders. 

I think it's better to manage the BOM in the code repository. I heard that Git is not efficient for Excel files. So I want to create a text file with formatting:
item , quantity , notes , example product link \n

Do you have better suggestions? I can work out the list during this weekend.

triffid, 2018-05-10:
CSV should work perfectly if you need a tabulated format

ikwic, 2018-05-18:
I'm currently buying some of the parts. And I have a question - is there any sort of tips how to distinguish the original servo MG92B from a fake?

borntoleave, 2018-05-18:
@ikwic  By luck? I did notice the variation in my separate small orders. Basically, the lower the price, the more chances you'll get fake ones. If I were to order the servos for my commercial kits, I'll contact the official manufacturer directly, order 1000+ units and ask for wholesale price. 
You can start to order the parts now, but note that I won't be ready to release more instructions in the following two months. You may use the components to test separate functional modules like moving servos, reading sensor data, etc. They will help you understand my later codes for driving the robot.

asmales, 2018-05-20:
Hi All,

I’m doing some building of my own, and printing PLA is the easiest for most people.
There is some good evidence that annealing PLA can produce the same strength (or better) as traditionally stronger printed plastics.
The caveat’s are is that is still has the same temp range, but in this application that’s okay and it shrinks the part being annealed, so the part has different physical dimensions.

The good news though, is that the shrink is predictable and repeatable – meaning that an oversized part can be printed and annealed to the final size.
The annealing for PLA is simply putting the part(s) on a baking tray in the oven at 55-65 degC. Then letting it cool (with parts in oven) – Remove parts when cool, Heat oven and Repeat, for an hour. At that temp there is no real softening, but the stresses in the plastic are transformed and the new shape (size) is produced.
Rigid.ink Quote:
“Through a relatively simple process, you can actually turn a standard material like PLA into one of the strongest 3D printer filaments… After heat treating PLA, you can expect to see some significant improvements in the strength of your PLA object. A 40% increase in strength and durability is not uncommon. Likewise, you can also expect to see a good improvement in stiffness. A 25% improvement here is not unexpected.”

Rigid.ink How to:
https://rigid.ink/blogs/news/how-to-anneal-your-3d-prints-for-strength

This is an example at 70 DegC for a few hours:
https://hackaday.com/2017/06/17/annealing-plastic-for-stronger-prints/

Machine Learning & AI
Additionally, there are new developments in image processing on the RaspPi using tech Intel has bought – Movideious.
https://www.movidius.com/

RaspPi example:
https://medium.com/deep-learning-turkey/a-brief-guide-to-intel-movidius-neural-compute-stick-with-raspberry-pi-3-f60bf7683d40

also this might be interesting,
Intel Real Sense Depth Cameras
https://www.mouser.co.uk/new/Intel/intel-realsense-camera-400/

https://www.mouser.co.uk/search/refine.aspx?Ntk=P_MarCom&Ntt=164888720 – the Video modules, not retail camera.

It’s possible to buy the Vision Processor D4 alone or on the processor card!
https://www.mouser.co.uk/search/refine.aspx?Ntk=P_MarCom&Ntt=129760865

Just keeping everyone informed.
Speak to you all soon.

Take care,
Adrian.

----------------------------------------
Adrian Smales BEng(Hons), MBA, MBCS
PhD Researcher: Health Informatics; Frailty

[/Users/Adrian/Library/Containers/com.microsoft.Outlook/Data/Library/Caches/Signatures/signature_1598266709]
Room C46
Edinburgh Napier University
School of Computing
10 Colinton Road
Edinburgh. EH10 5DT
UK

Mobile: +447768 034422
Email: a.smales@napier.ac.uk<mailto:a.smales@napier.ac.uk>
Research Homepage:  Adrian’s Page at IIDI<http://www.iidi.napier.ac.uk/c/people/peopleid/13375071>
LinkedIn: http://uk.linkedin.com/in/asmales
WebEx: https://<https://napier.webex.com/meet/A.Smalesnapier.ac.uk>napier.webex.com/meet/A.Smalesnapier.ac.uk<https://napier.webex.com/meet/A.Smalesnapier.ac.uk>
Twitter: @asmales
Skype: asmales

P Please consider the environment - do you really need to print this email?

“The reasonable man adapts himself to the world; the unreasonable one persists in trying to adapt the world to himself. Therefore, all progress depends on the unreasonable man.” - George Bernard Shaw 1856-1950, Irish Playwright and Critic

This message and its attachment(s) are intended for the addressee(s) only and should not be read, copied, disclosed, forwarded or relied upon by any person other than the intended addressee(s) without the permission of the sender. If you are not the intended addressee you must not take any action based on this message and its attachment(s) nor must you copy or show them to anyone. Please respond to the sender and ensure that this message and its attachment(s) are deleted.

It is your responsibility to ensure that this message and its attachment(s) are scanned for viruses or other defects. Edinburgh Napier University does not accept liability for any loss or damage which may result from this message or its attachment(s), or for errors or omissions arising after it was sent. Email is not a secure medium. Emails entering Edinburgh Napier University's system are subject to routine monitoring and filtering by Edinburgh Napier University.

Edinburgh Napier University is a registered Scottish charity. Registration number SC018373

borntoleave, 2018-05-22:
@asmales For most of the parts, using PLA is fine. The position of screw holes won't change much. Shrinking will be a problem for one or two structures where the mechanisms require a very small tolerance. I'm specifying ABS only because I have tested the dimensions after costly failures. For printing with PLA, some magnification factors has to be experimented and determined.

peterhyphenhyphen, 2018-07-08:
PLA is fine however PET, PETG and TPU are more common for parts requiring strength and durability. A useful tip from the 3D printing community is to use threaded brass inserts when needing to screw into a plastic part.  These are inserted using a soldering iron to heat the insert to place it by melting it into the body of the plastic. The brass will be durable and won’t strip steel screws. ABS is OK but warps and is relatively heavy compared to better plastics. There are dozens of filaments to try. I would get online and ask Tom Sanladerer for advice. Tom's YouTube channel is very educational and Tom is an engineer by training so understands the issues you face. https://youtu.be/CZX8eHC7fws Shows Tom testing many filaments including baked PLA for strength. Tom's filament tests are very thorough.

borntoleave, 2018-07-08:
Thanks for referring that resource! I will cite your note in later documentation on the instructions.

NirViaje, 2019-03-25:
Have you guys considered about K210 as the Machine Learning & AI processor? it's just $2.99 for the chip - 
https://kendryte.com/

my notes on K210 and relative things - 
https://github.com/NirViaje/nirviaje.github.io/issues/51

[image]
[image]

NirViaje, 2019-03-25:
> Hi All, I’m doing some building of my own, and printing PLA is the easiest for most people. There is some good evidence that annealing PLA can produce the same strength (or better) as traditionally [...]

may be chips like K210 low power consuming & low cost can work?

closed 2019-04-23 by borntoleave