PetoiCamp/OpenCat-Old #6: Suggestions for powering solution?

borntoleave, 2018-05-22:
I've been using 2S lipo batteries(7.4~8.4V) to directly power the servos. The servo's specification lists a maximal torque at 6.6V, but it works fine under 8.4V with some heating.

I hope to reduce the voltage to the safe 6.6V range. However when I connect some step down boards (9v to 6V, 3A max) in serial with the same 2S batteries, the output voltage will drop below 5V. It seems that the board will shutdown once peak current is higher than some threshold.

Do you know any cheap and neat solution to create a (6V, 5+ Amp) supply from the 2S lipo batteries? Thanks!

triffid, 2018-05-22:
How about something like https://item.taobao.com/item.htm?id=563344121202 ?

borntoleave, 2018-05-22:
It has a problem: too large in both size and input voltage.

I have tested several step down board options, there seems to be a gap between 3A and higher current. Below 3A, the circuit can be made as small as a fingernail. Higher current boards will at lease triple the size. Any insights on that? 
[3A board](https://www.ebay.com/itm/10pcs-Mini360-3A-DC-Voltage-Step-Down-Power-Converter-Buck-Module-3-3V-5V-9V-12V/132416658988?epid=1402279667&hash=item1ed4a5de2c:g:VrUAAOSw8SpbAdyM)
[5A board](https://www.ebay.com/itm/5A-XL4015-DC-DC-adjustable-step-down-module-buck-converter-4-38V-96/202216430664?hash=item2f1509dc48:g:6F8AAOSw-SBatssT)

From the raw voltage(7.4A to 8.4V) to target voltage(6.6V), there's around 1V drop. And the current will be 4A to 5A. The power formula gives P = V*I = 5W. I'm not sure how to regulate the voltage without generating too much heat.

triffid, 2018-05-22:
If 4A is enough, how about https://item.taobao.com/item.htm?id=546729706262
- just gotta change the feedback a little.

The seller would probably be willing to change them for you if you ordered
a bunch

STGRobotics, 2018-05-22:
I've been using the following to parts for my robots. they are small in size

Battery
Tenergy Li-Ion 7.4V 2200mAh Rechargeable Battery w/ PCB (2S1P, 16.25Wh, 4A Rate)
http://www.all-battery.com/li-ion1865074v2200mahrechargeablebatterymodulewithpcb.aspx
or
Tenergy Li-Ion 18650 11.1V 2600mAh Rechargeable Battery Pack w/ PCB (3S1P, 28.86Wh, 5A Rate, Samsung Cells)
http://www.all-battery.com/li-ion18650111v2600mahrechargeablebatterypackwithpcbandbareleads-31494.aspx

Regulator
Castle Creations CC Bec 10A 6S Switching Regulator
https://www.amazon.com/Castle-Creations-Bec-Switching-Regulator/dp/B000MXAR12/ref=lp_6179585011_1_1?srs=6179585011&ie=UTF8&qid=1526992164&sr=8-1

STGRobotics, 2018-05-22:
Tenergy sells whole for qty 100 or greater.
http://www.tenergy.com/

borntoleave, 2018-05-22:
@triffid Looks good. I‘ll order several for testing. Thank you!

STGRobotics, 2018-05-22:
Have you look at using Amazon for shopping cart and warehousing.

borntoleave, 2018-05-22:
@STGRobotics The kit will have tens of parts, and I have to preassemble some. I don't think Amazon can handle that for me.

closed 2019-04-23 by borntoleave