PetoiCamp/OpenCat-Old #7: License and STL files?

abetusk, 2018-10-22:
This project looks great, thanks for creating it!

I don't see any licenses on the source code or the STL (or other) files and the corresponding license files. The name "open" is in the name of the project and usually that has a connotation of being under an ["open source model"](https://en.wikipedia.org/wiki/Open-source_model) that makes source, design files and other materials available under a [free license](https://en.wikipedia.org/wiki/Free_license).

Is the intent that the source files and other design files be made under a free/libre license? If so, could you add the STL files along with a license to the code and other materials? If not, could you clearly explain in the documentation that this project is not open source?

borntoleave, 2018-10-22:
Open Source is more on the software side. The license will probably be MIT. If you check the codes in [TrivialTricks](https://github.com/PetoiCamp/OpenCat/tree/master/TrivialTricks), where I post some test codes, they should already been given MIT lisence. 

There will be at least one open source hardware model like https://youtu.be/uuVX23HES6g available after this busy launching month. I'll also take the last month to polish the codes until I'm happy with. 

Thanks for your patience!

abetusk, 2018-10-22:
Thanks for the response!

At the time I looked at the linked code, I don't see any source files with license headers, nor do I see 
a LICENSE file.

If you don't plan on releasing the STL or other build files, I would suggest stating clearly in the README or other documentation that these are not available and the "open" in "OpenCat" only refers to the source code. I imagine others have a similar expectation as I do that the "open" in the name refers to the source code and other files necessary to create the project, not just the software side.

Whatever you decide to release under a free/libre license, I would urge you to do so early without too much worry about cleaning the code. I've seen many projects claim to be "open", promising to release source or data files at a later date after they've been cleaned up, only to never be released. I don't believe this is due to malice but due to lack of attention from an overloaded schedule.

I believe publishing what you have and then adding as you update is a much better model. This allows the community to see what the process is, add if desired and, more importantly, makes sure that if for any reason you should abandon the project, the most recent snapshot is available to the public.

borntoleave, 2018-10-22:
There's no need to worry that I "promising to release source or data files at a later date after they've been cleaned up, only to never be released. " Nybble's motion is controlled by Arduino, and users will need to upload source code to play with it. Check back one months later when I start to ship the first Nybbles. 

My apologies that I just realized only a few of the codes are liscenced. I got that wrong impression because I did add MIT license in Nybble's driver code, and some explanations in the test codes. I'll add license to those test codes right after replying incoming emails. 

As to hardware, it's taking serious investments to build a working model. And I'm still trying to open it as soon as possible. The little yellow model in link was right designed for sharing. I was making some instructions but had to pause it for the campaign.

Fashim, 2018-11-01:
Hi borntoleave,

Hope you're doing well,
first and foremost I think this is an excellent project and hope to see it succeed.

However, I do have concerns over the open-source aspect of this project. By choosing to not publish the source code in it's entirety, you are doing yourself a disservice. FOSS software being fully transparent to those who want it allows for an entire community to help, and contribute to your project to make it greater than what's possible by a small team.

> By open-sourcing it, we will also face competetion[sic] from both large companies and copycat. We expect to release _most_ of the source codes with the launch of our first product.

From the README.md, this statement is in itself, an antithesis of the FOSS ideals and is concerning.

I understand that you have a financial commitment to this project and in doing so are expecting to either break even or make a return from selling the prefab kits.

As much as I'd recommend releasing the source code now it is ultimately your decision and if you choose not too, would it be possible to have a road-map of the release schedule for COTS BOM, STLs and source code?

Anyway happy programming!

borntoleave, 2018-11-01:
> Hi borntoleave,
>
> Hope you're doing well,
> first and foremost I think this is an excellent project and hope to see it succeed.
>
> However, I do have concerns over the open-source aspect of this project. [...]

Hi Fashim,
The Readme was written several months ago when people kept complaining about the "Open" in "OpenCat". 

Two days after our launch, a workable version for Nybble is already posted in the repository under **Nybble/** folder, with MIT license. And the code is compatible with the full version cat. I do need to add more comments to help understanding. 

My current priorities are getting all the hardware ready for backers. I‘m also filling up the wiki pages slowly. My goal is to have all orders fulfilled on time, and by that time an early tutorial is ready to use. I'm looking forward to in-depth discussions with you all!

borntoleave, 2018-11-01:
@Fashim 

The Readme was written several months ago when people kept complaining about the "Open" in "OpenCat".

Two days after our launch, a workable version for Nybble is already posted in the repository under Nybble/ folder, with MIT license. And the code is compatible with the full version cat. I do need to add more comments to help understanding.

My current priorities are getting all the hardware ready for backers. I‘m also filling up the wiki pages slowly. My goal is to have all orders fulfilled on time, and by that time an early tutorial is ready to use. I'm looking forward to in-depth discussions with you all!

lingfeng2, 2018-11-06:
Hi,great work！Can you give me a rough time about when you will add the stl or someting about hardware? I am a high school student from China. I can't wait to start my own open cat.Can you give me a rough time that when you will you? Thanks for you and your team again.

RongzhongLi, 2018-11-06:
Hi, @lingfeng2,
It's a hard question. At this stage Petoi is a hardware company just launched with [Nybble](igg.me/at/nybble) to get its initial funding, while the codes are already open to customers, researchers and potential competitors. It's impractical to share both hardware and software now. 

I'm also the only full-time labor to do research and handle production over the past two years. The challenges are far more than technical. If later you find me resume the development of the full version cat, that would be a sign that I've overcome my current obstacles and will be positive to share design files. 

For now I'd appreciate you read the codes and suggest improvements. I also think 3D printing directly from design files is no much fancier than printing a paper document. Buy a caliper and start to design the tiniest screw and nut. You will find the whole world is open source to you. 

授人以鱼不如授人以渔，thanks for your patience!

Breezor, 2018-11-06:
Agree that simply 3d printing the design files offers little.  However redesigning the files would advance nibble beyond the 1960''s archaic look and help to create a character that will go along with the advances in code.  The larger cat 3d printed looks far more appealing than the wooden nybble.  As a 3d designer and consumer the ability to choose physical characteristics  for a pet cat would be most beneficial and enticing for prospective pet owners.Regards,Brian GaleSent from my Samsung device

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/PetoiCamp/OpenCat/issues/7#issuecomment-436326650",
"url": "https://github.com/PetoiCamp/OpenCat/issues/7#issuecomment-436326650",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [PetoiCamp/OpenCat] License and STL files? (#7)",
"sections": [
{
"text": "",
"activityTitle": "**RongzhongLi**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@RongzhongLi",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"PetoiCamp/OpenCat\",\n\"issueId\": 7,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"PetoiCamp/OpenCat\",\n\"issueId\": 7\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/PetoiCamp/OpenCat/issues/7#issuecomment-436326650"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 399471110\n}"
}
],
"themeColor": "26292E"
}
]

RongzhongLi, 2018-11-06:
@Breezor I was hoping to design a wooden skull for camera too. But it seems easier to make a 3D printed one with standard mount to the tilt servo. Then it could switch on any design of heads.

Breezor, 2018-11-06:
Yes the head and face are most important to character.  Let me know if I can help you with that. Sent from my Samsung device

borntoleave, 2018-12-08:
@exilaus 
Wow! Impressive!
Have you checked the [head servo](https://github.com/PetoiCamp/OpenCat/blob/master/Resource/STL/mg90d.stl)'s STL model in the resource folder? 
If you are going to use the same servos in the kit, the dimensions would be very important.

exilaus, 2018-12-08:
know that infact that are a fork(have little issue with my git and i need reinstall it and after publish) . body little bigger for have more space for battery pack, anyway i waiting full set of servo but just test joint with some sg9 and md90s and work fine anyway cad have rotative joint for testing movement before cut. another point i want use only esp/esp32+ remote service have a nice day and good luck for your project!

zeroeth, 2019-03-15:
I think it would be great for those of us who want to customize the cat with acrylic parts of different colors if you added the vectors of the Nyblles design! Thanks! (PS thanks for making such a cute project)

borntoleave, 2019-03-19:
@zeroeth Acrylic will crack with self-tapping screws. Increase the screw holes, and use bolts+nuts instead. But it will make installation harder. Acrylic smells terrible under laser too. 

One day I‘ll reveal the design. Every slot and curve/fillet has a reason. For the time being, there are different [fan-made OpenCat](https://www.youtube.com/playlist?list=PLHMFXft_rV6OljaTr7zEM-f6do6Z0n5UG) now! 
I also have a fully DIY-able version, [CatMini](https://github.com/borntoleave/catMini) for my previous class.

zeroeth, 2019-03-22:
Respectfully, I think the point is if we back an open source project that includes hardware, and the hardware source is missing, we can't modify/extend/remix easily, just like with the software source.

Its not about the material/design choices you've made, those aren't reasons to not share the hardware of an open source project. And limits a whole other half of your audience being able to make and share cool ideas (on the hardware side)

Aiki-bot51, 2019-03-23:
I’m actually kind of disappointed, I’ve been following this project for 1 year+ now. Since the opencat video on YouTube. I learned about open source projects because of opencat, it is still the most interesting but when you funded nybble I thought “well maybe this is just a premium version on sold”, well I was wrong it’s actually the direction your going isn’t it? I’m mad and disappointed, but I understand your concern for rivals, because all your hard work doesn’t make money while other copycat makes money, but there is copyright law and fair use, but if I’m you I will probably do the same thing, but still a promise is a promise remember that! And you know what? Copy cat started appearing and their really open source

borntoleave, 2019-03-23:
@zeroeth @Aiki-bot51 
I've been using many open source software at school. But only until now could I publish one project and learn both the good and bad parts of it. For users, all they care about is the ease (and free) to use. But for the creators, their funding source is one key factor to decide their strategy. That's why there're various licenses, and open source doesn't necessarily mean open hardware or freeware.

R&D is expensive, especially hardware. The complexity of the original black cat is often underestimated. It took me three days to make one, and I only have one fully assembled at this moment. Here are some [samples](https://gallery.autodesk.com/fusion360/users/KT2ZR29VNX3H) of my model files. Very often more than three parts have to work together, collaborating with electronics and program. For the simplified Nybble, one iteration of PCBA alone takes 45 days and $1000. I'm running the fifth iteration now. Making the instructions for Nybble took two months, while it's just 1/5 as complex as the black cat. Operational cost of the company is ten times of R&D.

My passion is still in the black cat and I'm struggling towards it with achievable milestones. By developing NyBoard, it saves a half day in soldering. By making Nybble, people can start to understand the code structure and define communications between controllers. Nybble is just the beginning. I have tons of ideas that I had to give up job opportunities to follow. I'm 30 now, working 14/7 hours with $4 daily meals. And I have to leave this country after 9 years because I don't have a work visa. 

I encourage people to develop their own versions and I share their work. Simply releasing my design files does no good to innovation. Copy cat (who simply copies) don't taste the pain of trial and error. They don't have the dedication to keep the project sustainable either. I can't start a lifelong journey with a leaking tyre. Consider the demo you saw last year as an announcement of my pregnancy. The baby takes time to grow up. "A promise is a promise".

Breezor, 2019-03-23:
Bravo borntoleave!,  I understand the difficult journey you are on to try and develop a project with very limited resources.   For those who think you should give up everything (I was one) without recognizing the real obstacles that are holding you back from development, I say stand back and give the man room to work.   I wish I could support you more at the moment.  Go well and keep up the great work.

Aiki-bot51, 2019-04-14:
@borntoleave I’m sorry... i didn't know that you are still working hard to make it open source, it’s just for some of us in “unique country” buying stuff from outside has ridiculous taxes. For instance a nybble cost 230$ when it arrives I will need to pay 400-800$ which some of use still cannot afford. Do you use Patreon? I think many people will give you money if you can make sure that you will make it open source!

Aiki-bot51, 2019-04-14:
@exilaus can it be 3D printed? Your design. Can you post it on github please? Is it compatible with
[image]

borntoleave, 2019-04-15:
@Aiki-bot51 What country are you living in?

SG90 is analog plastic servo. I've tested but they break too fast.

Aiki-bot51, 2019-04-15:
@borntoleave indonesia. The taxes are crazy here (up to 300%) and 9g servo is about 1$ each. Its affordable the other kind of servo cost 5-8$ each( the es08md's ), its kinda out of my reach for now. I hope. With 9g servo i estimate i can make nybble around 50$. To clear any confusion im still a junior highschooler. So cannot afford premium especialy this is one of my first project that i thought i can make after following you for a few years.

borntoleave, 2019-04-15:
@Aiki-bot51 
Would the taxes be lower if imported from China?

The version on Thingiverse is not mine but can be a good DIY version to test on. 
The plastic 9g servo is not capable in both precision and strength. It will only be able to move in the air (without load). Check this [demo](https://youtu.be/B9_z5NMaN_E).

If it's your first project, I recommend you start with my simpler fully DIYable version [catMini](https://github.com/borntoleave/catMini).

Aiki-bot51, 2019-04-15:
@borntoleave no, the taxes wont be lower. And there are some corrupt people that likes to hold( i mean hold, is get it for themself ) the stuff if they're interested in e.g. phones, gadgets, cool stuff. Even tho we paid the tax full price.

I dont wanna do the catmini is because it doesnt look like a cat. Welp i guess i gotta save some money for better servo

Which one do you recommend? Es08mdi,es08mdii, or what?

nebrelbug, 2019-10-09:
@borntoleave any updates?

One thought I had is that you could possibly open-source just the Cat Mini (wouldn't need all of the features of Nybble) to both raise interest in Nybble and allow people to play with your code.

Norro21, 2020-08-26:
And despite saying 2 years ago that you would release it shortly, you have still not provided the derivative sourcecode and are in breach of the licences.

abetusk, 2020-08-27:
@Norro21, do you have proof of your claims?

The original owner of copyrighted material has every right to alter their work and not release it even if they provided a version of it under a 'viral' license (share-alike, GPL/AGPL etc).

I also don't remember seeing or hearing of any original documentation or other hardware digital artifacts that were released under a share alike license. The source code license involved are almost all MIT, even Adafruit's, which does not have a 'viral' clause, whose derived works can be kept proprietary.

Putting 'open' in a name of a project or giving a promise of releasing source code under a free/libre license is not a legally binding statement. I also don't know how much chastising people for not being more open will help.

borntoleave, 2020-08-28:
@Norro21 I'm trying to be as open as possible, but not too open to kill my company. Check this [video list of DIY OpenCat](https://www.youtube.com/playlist?list=PLHMFXft_rV6OljaTr7zEM-f6do6Z0n5UG). It shows many smart individuals were able to build from scratch with my already opened resources.

Norro21, 2020-08-28:
He stated it on his project page that he had to do it, and then has not followed through with what he is required to do. If he has used code with something like the GPL then he MUST share his derived work, it is not optional.

Yes people will recreate the project, but if you do opensource properly then their advances feed back into your project for you to benefit in the same way that you had a head start by using other people's opensource work. By closing it you don't stop them doing their own projects, you just force duplication of effort and ensure that community effort does not go back in.

You claim you want this to be a cheap development platform but pull the rug from underneath your potential customers by not providing the codebase for them to work on. Anyone's interest you peak has no choice but to duplicate your effort, leaving you out of the picture entirely.

borntoleave, 2020-08-28:
@Norro21 
Tell me, what is the following document and when it was published?
https://github.com/PetoiCamp/OpenCat/blob/master/Nybble/Nybble.ino

Is it C code or some French poem in your eyes?

Norro21, 2020-08-28:
But what I don't see is the nyboard which is derived from arduino?

borntoleave, 2020-08-28:
@Norro21 
> But what I don't see is the nyboard which is derived from arduino?

Now you are switching from blaming software to hardware. 

First, the "open-source" you familiar with are usually the software. I never claimed my robots are open hardware. 

Second, I already shared my block diagram two years ago, and it helped many people to make their DIY robots. 
[image]

Third, to be more accurate, our circuit board uses ATmega328P. It's been used on many other boards besides Arduino. We just decided to embrace the ecosystem of Arduino. Both parties are happy with the result. You can check this blog: https://blog.arduino.cc/2013/07/10/send-in-the-clones/ : Arduino derivatives and Arduino-compatibles.

Last, you cannot expect a young startup that barely breaks even to share both its core software and hardware after years of expensive R & D. It will kill the seed of innovation, only to make you and the copycats happy.

Norro21, 2020-08-29:
I can expect you to share it because YOU chose to include other people's work with licences that require you to share your derivative work.  You are the one killing innovation.

Breezor, 2020-08-29:
He has shared what is opensource by the basic standards and kept what is his own creation.

Norro21, 2020-08-29:
> He has shared what is opensource by the basic standards and kept what is his own creation.

Exactly and that is not acceptable. People shared their work with him on the condition that he shares his derived work. When he used their work he agreed to do that.

Breezor, 2020-08-29:
I don’t think it is derived.

Norro21, 2020-08-29:
He just posted the diagram showing the arduino that it incorporates, and has said in the past that it was based on it, so that seems a stretch.

Ultimately he's providing half the project, using his proprietary board as a paywall. He's not meeting the spirit of opensource while using it to kickstart to his development and as a selling point. By not meeting the licence conditions he is essentially stealing their work and it's a sorry state where he feels that he can't profit without ripping people off.

kaiaeberli, 2020-09-26:
Per https://opensource.com/article/17/9/open-source-licensing, it appears that the creators of the Arduino board could sue, but from above comment it seems they do not want to sue creators of such 'clones' (https://blog.arduino.cc/2013/07/10/send-in-the-clones/), even though they mention that these don't support the Arduino project.

Given in practice the license owner of Arduino does not mind, it does seem to go back to a philosophical point of whether it is more important to open source **everything** derived from an open-source license at any cost, or if this should be delayed until a firm financial foundation for the business is established.

In a sense: would we rather have someone creating these awesome toys and making robotics more accessible, or would you rather his business fails as he open sources too much of the critical components too early in the game or reinvents the wheel buy creating all code from scratch himself?

borntoleave, 2020-09-26:
Thanks @kaiaeberli !

[Arduino's official tweet sharing our product](https://twitter.com/arduino/status/1305875799686422528?s=20) may be the best answer to the concerns. We are working together to make cool projects more accessible to people. There're no conflicts between us. 

I initially built the prototype with an Arduino board. Later I got my friends designing the NyBoard, a customized PCB based on ATmega328P. We will release the BiBoard, another board base on ESP32. They both support the Arduino IDE but are not using Arduino's brand. There are many other projects built on those chips irrelevant to Arduino. Consider Linux: it's open-source but doesn't have a hardware product. What if they start to invent and sell computers? 

We are open-sourcing our codes but not publishing the hardware design. We don't forbid people recreating them for personal use, or sue tens of copycats who are selling our design. Creativity comes from a free and happy mind, not those mean and boring law terms.  

@Norro21 has been complaining and leaving negative comments everywhere since I debut the project. I simply didn't have organized codes to share at that time. I guess he never got a chance to check back or refresh his understanding of open-source.

Norro21, 2020-09-28:
@borntoleave I have not left negative comments 'everywhere'. I've complained once per project when you falsely complained it was open source.

While Arduino may not chase you on breaking their licence, the point you are conveniently ignoring is that if they had followed your strategy then your project would never have happened. By taking and not giving back you don't make these projects happen, you prevent something bigger happening, you kill a wider enthusiasm and creativity.  It is selfish, and totally against the ethos of sharing that YOU RELIED ON TO CREATE YOUR PROJECT.

borntoleave, 2020-09-28:
@Norro21 
Someone saw a tree and took picture A with his camera. You like the picture, so you found the tree and took another picture B with your camera. Who owns the copyright of picture B?

In our case, Arduino used the datasheet of ATmega328P to make the Arduino Uno (**hardware**). I first saw Arduino Uno, then developed our NyBoard (**hardware**) with the datasheet of ATmega328P. We found Arduino IDE (**software**) was convenient so we adopted it on our board and open-sourced our OpenCat code (**software**). 

When announcing the OpenCat project, it was a rough prototype with some working codes. Later we made the PCBs, Nybble, and Bittle (**hardware**) and worked hard to made them affordable. They are running with the OpenCat code (**software**), which is open-sourced here. There're no false claims. 

You could come up with a more precise and true claim and try to shrink it into one sentence. You can also ask people who used our shared resources to make their robot pets. For your convenience, here is the [YouTube playlist of DIY OpenCats](https://www.youtube.com/playlist?list=PLHMFXft_rV6OljaTr7zEM-f6do6Z0n5UG).

zeroeth, 2020-11-06:
You've done it again with Bittle. Presenting it as an open source project, but its a closed hardware platform plain and simple. The hardware design files are not available. Clever wording like:

'Bring down the technical and cost barriers of dynamic quadruped robot to an open-source pet'

and

'Evolving with an open-source gene, Bittle is an open platform to fuse multiple makers' gadgets into one organic system.'

Nybble has no design files available. Bittle has no design files available.

Stop trying to present these open source projects.