PetoiCamp/OpenCat-Old #11: Suggestion port to PIC32

cb88, 2019-03-29:
PIC32MX and PIC32MZ MCUs are vastly faster than AVR, while retaining almost all of the ease of programming via Chipkit's Arduino framework and since Microchip has bought Atmel it is the same company anyway.

A 40Mhz PIC32MX270 chip is very similar to an Arduino Leonardo (see the Chipkit Lenny) but offers much stronger performance and native 32bit instead of 8bit. There is also the MX274 80Mhz chip but it isn't well supported yet by Chipkit. Some of the MZ series chips are as fast as 250Mhz and up to 32MB of DDR on package.

I've developed some things for work and the additional speed is amazing things that too several ms on an AVR can be done in uS on a  PIC32 for about the same price.  

Another Idea is to use the IRDA UART built into PIC32 to allow the cats to talke to each other and some IR devices via IRDA.

I think this could allow the balancing algorithm to run a lot faster and work better than on AVR perhaps and with less coding effort to make it fast on a slow MCU.

Dobloz, 2019-03-30:
This sounds a great idea!
Are you considering building the Nybble incorporating a PIC32 ?
Please share!

borntoleave, 2019-03-30:
Yes, that's one plan for future upgrades. NyBoard V0 is designed to have a dial switch to accept faster chips as the master for I<sup>2</sup>C peripherals. It's quite open and convenient to add other chips. I've already tested with SAMD21 and ESP32 on individual code modules. 
The two chips are: 
(48MHz clock, 32KB SRAM, 256KB flash ) compared with the current ATmega328P's:
(20MHz clock,   2KB SRAM,   32KB flash)
The motion will be like upgrading from 24 fps to 60 fps HD movies!

Faster chip will boost balancing and overall performance. However, that will also lead to faster wearing of the hardware, and require higher current to drive both the chip and servos. It will require heavier batteries, and introduce more burden on the servos. So the new system has to be thoroughly evaluated to get a balanced outcome. As a result, even the material cost would be easily doubled/tripled than the current configuration of Nybble. Not to mention the current PCB is already too crowded. 

There will also be some restructuring of the code. When designing Nybble, I considered many DIYers who only have Arduino Uno in hand. So I kept the usage of on-board EEPROM. I also deliberately compressed all the codes into a "prehistoric" chip to show the simplicity/efficiency of OpenCat. But for the next upgrade, most of the constants will be saved to SRAM directly, making it easier for 3rd party developers.

cb88, 2019-03-30:
MikroElektronika Flip&Click MZ  uses an PIC32MZ2048EFH100 ($12) and is supported by Chipkit toolchain, and provides 2MB flash  512kB sram 75 digital IO  10 analog inputs 5 uarts  3 spi  3 i2c ports, and comes in a TQFP-100 package so not too hard to solder... I find that sometimes going for a bigger chip ends up reducing complexity in the end as you don't need to use IO expanders or have easier routing because you have more IO available on the chip. Another interesting chips is the K210 which can do 8 channel audio input, and also can interface with a camera for computer vision https://www.analoglamb.com/product/sipeed-maix-1-w-risc-v-dual-core-64bit-fpu-wifi-ai-module-core-board-development-board-mini-pc/

Wear on the servos would only increase if the rate of updating motions increased.... which you wouldn't *have* to do, using a faster MCU is more about increasing capability beyond just locomotion and basic sonar, without adding the complexity of an embedded PC like RPi.