PetoiCamp/OpenCat-Old #12: Repetitive information in serial monitor at start

borntoleave, 2019-04-23:
The following text is one example output in the serial monitor:
```
⸮
* Starting *
Initializing I2C
Connecting MPU6050...
Testing connections...
MPU successful
Initializing DMP...
794 148 19 61 
Enabling DMP...
Enabling interrupt detection
DMP ready!
⸮
* Starting *
Initializing I2C
Connecting MPU6050...
Testing connections...
MPU successful
Initializing DMP...
794 148 19 61 
Enabling DMP...
Enabling interrupt detection
DMP ready!
```
I'm not sure where the ⸮ symbol comes from. And the sketch may start twice at the beginning. Any ideas? Maybe it's related to the timing around Serial.begin(). That's why I'm having `  Serial.setTimeout(5);` or
`while (Serial.available() && Serial.read());`
for clearing the serial buffer. 
But the "⸮" and duplicated/reboot information still appear.

Any ideas?

dahchael, 2019-04-23:
?

kaiaeberli, 2020-09-26:
Maybe it is to do with the baud rate of the Serial.begin() and Serial monitor not being the same? See: https://forum.arduino.cc/index.php?topic=528920.0.