PetoiCamp/OpenCat-Old #13: On Authorization

zhuangfengzi, 2019-04-24:
Is this project authorized?

If we use this open source code for simple development, such as Bluetooth version or other, to make a basic mechanical cat, can we manufacture and sell it?

Aiki-bot51, 2019-04-24:
I dont think so.... And you shouldnt rhongzhong li has worked hard gor this code. Maybe try distributing his nybble

zhuangfengzi, 2019-04-24:
@Aiki-bot51 ，
I read the topic “License and STL files?”, did not understand!
That's why I asked this question.

Aiki-bot51, 2019-04-24:
For instance there is a copyright on the nybble.ino code @zhuangfengzi
If you can do 3d modeling please contact me on aikyodzaki@gmail.com

zhuangfengzi, 2019-04-29:
@Aiki-bot51
What can I say here in the future?

The happiest thing is that Professor Li will share it all now!

My main purpose of studying is to improve my life. I can hardly eat any food now. If you can't earn money, what else can you do?

What you learn must always be translated into value! The most vulgar way to say it is money!

closed 2019-04-29 by zhuangfengzi

borntoleave, 2019-04-29:
Hi, 
Sorry for late response. I'm fried working on sending out orders. Will return to you with my vision on this project.